MrAkergren/SolarRemote #69: Duplicate memory card

MrAkergren, 2015-02-25:


MrAkergren, 2015-02-27:
Done when:

* Operating system installed
* LXDE uninstalled
* Python3 installed
* SerialPythinstalled
* PyTFT - scripts
* copied .xsession

straaand, 2015-02-27:
Have some problems getting the GUI autostarting and rotate the screen

MrAkergren, 2015-02-27:
The rotation should be 180 degrees, also a config file should altered where "invertYaxis: true" should be entered, that should fix the rotation. I'll investigate more tomorrow

MrAkergren, 2015-02-28:
The file: ``` /etc/modprobe.d/adafruit.conf  ```
 should have the argument  ``` rotate=180  ```  for 180 degrees rotation and the file

 The file
``` /etc/X11/xorg.conf.d/99-calibration.conf ``` 
need the arguement 
``` Option      "InvertY"       "true" ``` 
to get the axis right.

straaand, 2015-03-06:
Auto login: http://www.opentechguides.com/how-to/article/raspberry-pi/5/raspberry-pi-auto-start.html

MrAkergren, 2015-03-06:
added ```~/.xinitrc```with command 
``` bash
exece python /home/pi/solarremote.py
```

and  confed `~/.profile` to end with

```bash
# if logging into tty6 (which will autologin), run startx
if [ -z "$DISPLAY" ] && [ $(tty) = /dev/tty6 ] ; then
    startx ;
fi

djqfr, 2015-03-13:
Is this done?

straaand, 2015-03-13:
The original issue was about duplicating the installation on my memory card but I think it changed to creating a *.img of the setup so you just have to flash it to a new memory card.  Oskar did a disk clone but not sure if it was usable?

MrAkergren, 2015-03-13:
this issue should be closed, see first comment for what it covers. For backup, see issue #79